esp-rs/esp-flasher-stub #24: Use AT-CMD feature of the uart to detect end of slip packets

MabezDev, 2023-07-13:
Currently, the uart is being interrupted on _every byte_ this is not optimal for performance, and especially at higher bauds we make miss single bytes. We should use the AT-CMD feature to detect end the end of slip packets and fetch a packet at a time in the interrupt.

igrr, 2023-07-13:
Alternatively, we could rely on rxfifo_full_thrhd to raise an interrupt only when some moderately large number of bytes (e.g. 100 bytes) are in the FIFO? That would still gave enough time for the interrupt handler to copy those bytes out of the FIFO before an overflow happens.

MabezDev, 2023-07-13:
Yeah, I think that would work better actually, because our slip packets are quite small.

DNedic, 2023-07-17:
@igrr How would this work when the packet exchange ends and there are say 90 bytes left in the FIFO?

igrr, 2023-07-17:
The simplest option is probably to enable the RX timeout interrupt. It will fire if no byte has been received for a given number of cycles.