esp-rs/esp-flasher-stub #37: ESP32-S3 fails to flash when using the UART Port

SergioGasquez, 2023-11-28:
Flashing the ESP32-S3-DevKitC-1 fails when using the UART Port with: 
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `espflash flash --monitor target/xtensa-esp32s3-none-elf/debug/s3`
[2023-11-28T08:12:51Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-11-28T08:12:51Z INFO ] Connecting...
[2023-11-28T08:12:51Z INFO ] Using flash stub
[2023-11-28T08:12:52Z WARN ] Setting baud rate higher than 115,200 can cause issues
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ChangeBaud command
```
- Using `--no-stub` we can flash it using the UART Port with no issues
- Using the USB port with the stub also works fine.

Using `espflash` with latest changes on `main` ([`0420d79`](https://github.com/esp-rs/espflash/commit/0420d79a4ff8cf9aafcaabc38952963ed46ba582)).

jessebraham, 2023-11-28:
I've also experienced intermittent timeout issues with some chips. Sorry I have not made notes of this, I will try to collect some data.

jessebraham, 2023-12-04:
I can confirm issues flashing the C6 via UART as well. Thanks for the info, we'll look into this.

jessebraham, 2023-12-06:
I've updated the flasher stub to use the latest `main` branch from `esp-hal`, and this has at least fixed my connection issues for the ESP32-C6. I will do some more testing to verify that things are working across the board.

I will open a PR with my changes when I have confirmed things are all working.

PR #39 (2023-12-22): Update to latest HALs, various fixes/cleanup
Still needs some more thorough testing, but at least for me this resolved the UART issues with the ESP32-C6 and ESP32-S3. We will need to test the other devices and the USB Serial JTAG too, I suppose, so will leave this as a draft for now.

Closes #37 (I hope 😁)

(Sorry about the noise in the TOML files, had forgotten to disable formatting for them and was too lazy to revert the changes 😅)
closed 2023-12-22 by jessebraham