esp-rs/esp-flasher-stub #45: espflash::timeout on esp32c6 via JTAG uart

t-moe, 2024-01-22:
I'm using espflash master to flash a esp-idf-* (master) project with espflash to a esp32c6.

```
espflash flash --monitor <ELF>
[2024-01-22T08:43:49Z INFO ] Serial port: '/dev/ttyACM0'
[2024-01-22T08:43:49Z INFO ] Connecting...
[2024-01-22T08:43:49Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ReadReg command
```

Some more debug output of espflash

```
[2024-01-22T08:43:36Z DEBUG] Finish stub write
[2024-01-22T08:43:36Z DEBUG] Writing command: MemEnd { no_entry: false, entry: 1082130432 }
[2024-01-22T08:43:36Z DEBUG] Stub written!
[2024-01-22T08:43:36Z DEBUG] Writing command: ReadReg { address: 1073745920 }
Error: espflash::timeout
```

I've recompiled the stubs with dprint enabled, and I get the following output from the stub:

```
Stub init! Transport detected: UsbSerialJtag
Stub sending greeting!
Waiting for command
```

I've also slightly modified src/flasher/mod.rs (espflash)

```
         debug!("Stub written!");
 
         match self.connection.read(EXPECTED_STUB_HANDSHAKE.len())? {
             Some(resp) if resp == EXPECTED_STUB_HANDSHAKE.as_bytes() => Ok(()),
             _ => Err(Error::Connection(ConnectionError::InvalidStubHandshake)),
         }?;
 
+        debug!("Handshake ok");
+        std::thread::sleep(Duration::from_secs(3));
+
         // Re-detect chip to check stub is up
         let magic = self.connection.read_reg(CHIP_DETECT_MAGIC_REG_ADDR)?;
         let chip = Chip::from_magic(magic)?;
         debug!("Re-detected chip: {:?}", chip);
 
         Ok(())
```

It seems the handshake is written, but then the chip redetect fails.

When I flash the same binary via Uart0 instead of USB JTAG, it works...

SergioGasquez, 2024-01-23:
I've tried reproducing this error with no success, maybe is something related with HW? Could you try using a different cable? 

`espflash` is sending the command to detect the chip, but stub is not receiving it for some reason…  and this seems to happen with C stubs (are the ones being used in espflash 2.1) and Rust stubs (being used in espflash main), so Im not really sure if this issue belongs here now.

Have you tried with esptool?

t-moe, 2024-01-29:
Same here:
Thanks for trying to reproduce this @SergioGasquez .
I can also not reproduce this reliably.  There are times where this issue occurs constantly and then there are times where it does not occur for hours/days. 
I'll keep this issue open for a while and try to add more info, try with esptool + different cables, as soon as the issue occurs again.

MabezDev, 2024-01-29:
It could be a similar error condition we reach in probe-rs: https://github.com/probe-rs/probe-rs/issues/1963#issuecomment-1913575783. Essentially a USB-JTAG-SERIAL reset isn't a real reset and some parts of the chip are left running, one of these could be the memory protection peripheral. Though, at first glance I can't see any similar workarounds in esptool.py 🤔.

turulix, 2024-08-10:
Im facing the same issue in `espflash v3.1.0` where flasing apparently works with `--no-stub`  but sometimes failes without it. It used to work fine in the beginning and then just randomly stopped working. Flashing with `esptool` works fine without any issues.