esp-rs/esp-flasher-stub #49: Flashing with baud rate higher than 115,200 causes a timeout in `v3.0.0-rc.1`

AnthonyGrondin, 2024-03-04:
I'm running the same flash command, with `--baud 921600`, and it used to work previously, (IE in v2.1.0) but it is now returning a timeout error.

**With v2.1.0 (expected result)**:
```log
     Running `espflash flash --port /dev/ttyUSB1 --baud 921600 --monitor target/xtensa-esp32s3-none-elf/release/my_app`
[2024-02-27T19:59:58Z INFO ] 🚀 A new version of espflash is available: v3.0.0-rc.1
[2024-02-27T19:59:58Z INFO ] Serial port: '/dev/ttyUSB1'
[2024-02-27T19:59:58Z INFO ] Connecting...
[2024-02-27T19:59:59Z INFO ] Using flash stub
[2024-02-27T19:59:59Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       7c:df:a1:e2:5a:0c
App/part. size:    982,864/8,323,072 bytes, 11.81%
[00:00:00] [========================================]      14/14      0x0                                                                                                                        
[00:00:00] [========================================]       1/1       0x8000                                                                                                                     
[00:00:15] [========================================]     599/599     0x10000                                                                                                                    
[2024-02-27T20:00:16Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 
...
```

**With v3.0.0-rc.1 (bug: timeout)**:
```log
     Running `espflash flash --port /dev/ttyUSB1 --baud 921600 --monitor target/xtensa-esp32s3-none-elf/release/my_app`
[2024-02-27T20:01:50Z INFO ] Serial port: '/dev/ttyUSB1'
[2024-02-27T20:01:50Z INFO ] Connecting...
[2024-02-27T20:01:51Z INFO ] Using flash stub
[2024-02-27T20:01:52Z WARN ] Setting baud rate higher than 115,200 can cause issues
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ChangeBaudrate command

...
```

SergioGasquez, 2024-03-04:
Moved this issue from `espflash` repo. Using `espflash 3.0.0-rc.1` with `--no-stub` or the C stub works fine.

SergioGasquez, 2024-03-04:
- I've tried regenerating the S3 stub:
  - Using esp-hal `main` (`rev 3b6a3cf3a4215889f803e3c1b9e684e2248e2db5`) with the following changes:
  ```
  diff --git a/esp-hal/src/uart.rs b/esp-hal/src/uart.rs
  index 08bd24c..ba32845 100644
  --- a/esp-hal/src/uart.rs
  +++ b/esp-hal/src/uart.rs
  @@ -799,13 +799,13 @@ where
       fn change_baud(&self, baudrate: u32, clocks: &Clocks) {
           // we force the clock source to be APB and don't use the decimal part of the
           // divider
  -        let clk = clocks.apb_clock.to_Hz();
  +        let clk = clocks.xtal_clock.to_Hz();
           let max_div = 0b1111_1111_1111 - 1;
           let clk_div = ((clk) + (max_div * baudrate) - 1) / (max_div * baudrate);
   
           T::register_block().clk_conf().write(|w| unsafe {
               w.sclk_sel()
  -                .bits(1) // APB
  +                .bits(3) // 1: APB, 2: RTC_FAST, 3: XTAL
                   .sclk_div_a()
                   .bits(0)
                   .sclk_div_b()
  ```
  - In  `esp-flasher-stub`  I just pointed to that `esp-hal` repo using [`esp-hal-upgrade` branch](https://github.com/esp-rs/esp-flasher-stub/tree/esp-hal-upgrade) changes.
  - This could be useful since we read the `UART0_CLKDIV_REV` in `get_uart_div()`

- Not sure if we should `send_greeting()` in the `ChangeBaudrate` processing.
  - Looking at [`ESP_SET_BAUD` in the C stub](https://github.com/espressif/esptool/blob/master/flasher_stub/stub_flasher.c), it does not send the `OHAI` message
- C3 works fine:
  - Printing the received response in `espflash` of the ChangeBaudrate command results  in:
    ```
    [2024-03-04T14:54:29Z WARN ] Setting baud rate higher than 115,200 can cause issues
    ...
    CommandResponse { resp: 1, return_op: 15, return_length: 2, value: ValueU32(0), error: 0, status: 0 }
    ```

SergioGasquez, 2024-03-04:
Important update: Seems like the issue does not affect all the S3 targets. Some folks are able to flash different S3 devkits while changing the baudrate. Even the same devkit (`ESP32-S3-DevKitC-1 v1.0`) works for some people and does not work for others.

I've also tested using different wires/computers and the results, for me, are the same: 
- ESP32-S3-DevKitC-1 v1.0 does not work
- ESP32-S3-BOX works fine

SergioGasquez, 2024-03-05:
Hi @AnthonyGrondin, I think I found the issue, could you try the following branch and see if it solves the issue for you?
```
cargo install espflash --git https://github.com/SergioGasquez/espflash --branch fix/s3-stub --force
```

AnthonyGrondin, 2024-03-05:
I've tried with the updated espflash `origin/main` and it fixes the issue. Thanks for taking care of this.

closed 2024-03-05 by AnthonyGrondin