esp-rs/esp-flasher-stub #54: Add support for ESP8266

dobairoland, 2024-03-15:


SergioGasquez, 2024-03-15:
See https://github.com/esp-rs/esp-flasher-stub/issues/36

jessebraham, 2024-03-15:
> To the best of my knowledge the goal is to use esp-flasher-stub in all Espressif SW products where stub flasher is used, including esptool.

This is news to me, it would be nice if this could be better communicated. I was told that it _might_ be used in one or two flashing applications ~2 year ago, and we have received little to no feedback since, and definitely no assistance with developing the stub.

> However, if esp-flasher-stub won't support ESP8266 then we will have to keep maintaining two flasher stubs which kills the main idea of to have just one flasher stub.

I empathize this, but I don't understand how we're expected to add ESP8266 support when we don't have a working HAL for it. We have a small team and already we barely have the resources to keep up with the projects we already have. The ESP8266 is EOL in ~2 years and this does not feel like a good use of our time.

Maybe somebody from another team can step up and make this happen if it is a priority.

closed 2024-04-16 by dobairoland