esp-rs/esp-flasher-stub #63: invalid stub handshake response received

sakuwa, 2024-05-31:
Hello
i got the following issue:
```
[2024-03-25T15:55:55Z INFO ] Serial port: 'COM5'
[2024-03-25T15:55:55Z INFO ] Connecting...
[2024-03-25T15:55:57Z INFO ] Using flash stub
Error:   x Error while connecting to device
  `-> Invalid stub handshake response received

```

i can connect with py -m esptool -p .... to my sparkfun esp32 thing board.

I use this instruction guide to flash the device: https://github.com/beeb/coffee-scale-app/blob/main/rs/README.md

Also i set the baudrate to a lower one (115200) in the espflash.toml without success. 

I tried to find with google the issue with the handshake and thats why i set the baudrate to a lower part. 
What did i wrong or miss?

beeb, 2024-06-14:
Hey @SergioGasquez sorry for the late reply.

Still no luck with espflash 3.1.0. Getting a `Error: espflash::timeout` when I don't pass the `--before no-reset` option, and I get a `Error: espflash::connection_failed` when I pass the `--before no-reset` option.

beeb, 2024-06-14:
Ok I updated cargo-espflash to 3.1.0 and the error is slightly different:

without option:

```
Error:   × Error while connecting to device
  ╰─▶ Invalid stub handshake response received
```

with `--before no-reset`:

```
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
```